publiclab/pi-builder #1: Build scripts for Lego spectrometer + Basic Microscope

jywarren, 2018-03-27:
We're doing a build script that will:

* start with Raspbian
* install the PHP-based https://elinux.org/RPi-Cam-Web-Interface for accessing a Raspberry Pi camera
* use [these config files](https://gist.github.com/jywarren/ce82d8c6ec3a05e591b7767d1a33c176) to broadcast a WiFi network called `00-PiCamera`
* install `raspap-webgui` - to administer -- via https://gist.github.com/jywarren/ce82d8c6ec3a05e591b7767d1a33c176#file-rasberrypi-txt-L38-L56 
* change `/etc/hosts` and `/etc/hostname` to a unique name so more than one can be run at a time (from a list of animal names)

itdaniher, 2018-03-28:
The last part - unique names, is probably the only part that you'll need to actually implement in custom software. The rest looks like configs and a bunch of systemd unit files / networkmanager configs. You'll probably want to interrogate the MAC address or SD card UID, do some bitmath, index a list of names on first boot to set hosts / hostname.

jywarren, 2018-04-01:
Here's an edited version of our `.bash_history` I'm trying to narrow down into a build script. Thanks, @itdaniher !!

https://gist.github.com/jywarren/48bf4d5ef1d89d0e74472cf8a46f8154

jywarren, 2018-04-01:
https://github.com/Drewsif/PiShrink

jywarren, 2018-04-01:
OK, this went OK... but i got a lot of `NOPASSWD` notices for the copying steps... i should just remove sudo

jywarren, 2018-04-02:
Ok! Seeing this error now: 
[image]

jywarren, 2018-04-02:
Great - this is looking OK, although I haven't been able to get it to start up a WiFi network. I've been refining the install script, and named it `inside-setup.sh` to indicate that it's to be run inside the Pi, and not on the `.img` from outside. 

By contrast, https://github.com/Drewsif/PiShrink should be run on the image from the outside.

jywarren, 2018-04-02:
Not all of the template config files seem absolutely necessary -- a couple I might not have actually changed. But I wanted to be thorough! 

I think I need to figure out why `ifup: failed to bring up wlan0` is appearing in the readout, but I'm pretty close here.