publiclab/pi-builder #2: Reproducible process from default rapsbian.img => buildScript.sh => modifiedRaspbian.img

jywarren, 2018-03-27:


itdaniher, 2018-03-28:
Not sure if this is a general RFC or not, but I've had a lot of luck with an ansible playbook modifying a default (raspbian/armbian) image mounted as a chroot. With a little work to clean up timestamps, it'd likely be reproducible. I've found it to be a very high quality solution to building custom flashable images for embedded linux boards.
happy hacking!

icarito, 2018-03-28:
Thanks Ian we really appreciate the comment.

I think @jywarren doesn't mean strict _binary_ reproducible builds, but something that is automated.

@itdaniher do you have some recipe to point us? I have no experience with Pi _or_ Ansible - so it will be fun.

itdaniher, 2018-03-28:
I'll write up a bit about the workflow later today, happy to help publiclab. Ansible is very easy to learn and very well documented.

Only caveat is that getting cross architecture chroots working is a bit of a pain, so it's best to build images on a compatible architecture. Scaleway.com, a rock64, or maybe even a pi3 itself would work nicely as a beefy ARM buildbox.

jywarren, 2018-03-28:
THANK YOU! Yes, we're putting this in a repo partially to "ask the world" for ideas on this. I also think @rjsteinert may have thoughts about this one? I remember Open Pipe Kit had prebuilt `.img`files for Pi -- what do you think, RJ?

itdaniher, 2018-03-28:
Here's a very basic example of an ansible playbook making modifications to prepare an embedded Linux image for our use:

https://gist.github.com/itdaniher/14d0981c028022ed7fd2da018ff24296

Also need to mount the image before you can chroot into it, a quick google[1] suggests a command like

```sudo mount -o loop,offset=62914560 raspbian_image_file_here.img /myfolder```

will work. Armbian uses a different offset (4194304) and I'm not sure that 62914560 is correct for the latest raspbian.

then you put the following in a file called "hosts"

```
[raspbian_chroot]
/myfolder ansible_connection=chroot
```
and can run something like:

```
sudo ansible-playbook -i hosts -l raspbian_chroot dev.yml
sudo unmount /myfolder
```

and flash your image!

We have a 128 line "production" playbook that does a variety of tasks like hardcoding an ssh public key, blacklisting some kernel modules, installing some systemd services, changing the SSH listening ports, etc, but I think it solves problems you don't need solved.

Pluses of this route: ansible is a pretty well-known automation tool, shouldn't be hard to recruit volunteers to maintain your application. It's really well documented. [2]

Biggest downside: PITA to chroot from x86 into an ARM environment, at least it was last time I tried. You have to run Ansible and mount your image on an ARM CPU - a Pi3 should work with multiarch support.

[1] https://www.raspberrypi.org/forums/viewtopic.php?p=380844
[2] for an example, see http://docs.ansible.com/ansible/latest/modules/lineinfile_module.html

jywarren, 2018-04-01:
I'm wondering if the chroot would work on a `crouton` enabled chromebook (https://github.com/dnschneid/crouton) in case that's easier to find.

jywarren, 2018-04-02:
I tried running our script on a chrombook chroot but the 16.x xenial ubuntu isn't close enough to Raspbian for it to completely work. Interesting though, i'm sure some virtualization could help but was curious.

icarito, 2018-04-02:
Here's some information suggesting it's possible to access an ARM chroot with qemu:
https://unix.stackexchange.com/questions/41889/how-can-i-chroot-into-a-filesystem-with-a-different-architechture

icarito, 2018-04-02:
Jeff, I was able to mount and virtualizedly chroot into it from tycho, following the reference I just shared:
```
# This provides the qemu-arm-static binary
apt-get install qemu-user-static

# Mount my target filesystem on /mnt
mount -o loop fs.img /mnt

# Copy the static ARM binary that provides emulation
cp $(which qemu-arm-static) /mnt/usr/bin
# Or, more simply: cp /usr/bin/qemu-arm-static /mnt/usr/bin

# Finally chroot into /mnt, then run 'qemu-arm-static bash'
# This chroots; runs the emulator; and the emulator runs bash
chroot /mnt qemu-arm-static /bin/bash
```
[image]

icarito, 2018-04-02:
in order to mount the img that has two partitions, I used kpartx -a sdcard.img.cdr first, this created a device mapper file under /dev/mapper/

icarito, 2018-04-09:
I looked for the terms 'raspberry' and 'kpartx' and found quite interesting projects:

- https://github.com/munnerz/rpi-builder-docker
- https://github.com/Fl0-0/NFC-Box
- https://github.com/hypriot/image-builder-rpi

@jywarren you mentioned we used a custom image to start with, from parts and crafts?

I have the one uploaded to Tycho but for this to be _reproducible_ in the medium term we will need to be able to build newer versoins.

icarito, 2018-04-09:
Oh I see the Parts and Crafts image building process is based in this: https://publiclab.org/notes/partsandcrafts/12-03-2017/setting-up-the-raspberry-pi-camera-for-wireless-streaming - correct?

jywarren, 2018-04-09:
> setting-up-the-raspberry-pi-camera-for-wireless-streaming - correct?

yes, exactly! wow this is cool. Do you need anything else at the moment?
The script I wrote in this repo is designed (for the time being) to run on
the Parts & Crafts .img in their guide. I did it and then see a `wlan0`
error.

rjsteinert, 2018-04-10:
HI @jywarren - I'm a little late the thread but FWIW on the Open Pipe Kit R-Pi images we started with manually building them on the hardware they were intended and then did experiment with using qemu but it was just becoming possible back then.  That would be great if it's not too painful now, otherwise a bash script and manually building images just gets the job done.

jywarren, 2018-08-03:
Hi, all - just wanted to chime in that @icarito has completed a large portion of this! You can now read the manual in the README: https://github.com/publiclab/image-builder-rpi

Fantastic!!!

icarito, 2018-08-03:
Thanks @jywarren ! I hadn't pushed the manual yet, but it's published now!

https://github.com/publiclab/image-builder-rpi/releases/download/0.1/manual.pdf

Sources are in [docs/](https://github.com/publiclab/image-builder-rpi/tree/master/docs).

jywarren, 2018-08-03:
oh sorry, i thought it was there, my bad! But good to see now :-)