publiclab/pi-builder #3: virtualized testing for Raspberry Pi

jywarren, 2018-05-01:
This site lists RPi as a continuous integration target... in wonder how it worked? (link seems broken...)

https://github.com/liuliu/ccv

jywarren, 2018-05-01:
Also some good leads here:

* https://www.reddit.com/r/devops/comments/6zwrb1/implementing_continuous_integration_with/
* Amazon AWS apparently now has something? https://www.reddit.com/r/raspberry_pi/comments/6zrwrd/amazon_ec2_systems_manager_adds_raspbian_os_and/
* Raspberry Pi rackmount rentals: https://www.mythic-beasts.com/order/rpi
* https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/
* http://thesociablegeek.com/node/github-continuous-deployment-to-a-raspberry-pi/

icarito, 2018-07-24:
Do you mean to mount a built image and run tests on it? What kind of tests do you have in mind?

jywarren, 2018-07-24:
Yeah, like, build an image, mount it, then run an automated test suite that confirms that a given script worked, so looking for things like:

* wifi hotspot installed and online
* correct utilities available and basic functionality without errors
* (ideally) can take a photo and confirm that it's not an empty file?

Stuff like that - what do you think?

icarito, 2018-07-24:
I think you are going to like what I found!

I noticed the end of the [output](https://jenkins.laboratoriopublico.org/job/Raspberry%20Kit%20Image/13/console) of the image building process:

```
+ rspec --format documentation --color /workspace/builder/test

SD card image
  exists
  Partition table
    has two partitions
    has a boot-partition with a vfat filesystem
    has a root-partition with a ext4 filesystem
  Binary dpkg
    is compiled for ARM architecture
  /etc/fstab
    has a vfat boot entry
    has a ext4 root entry

Root filesystem
  is based on debian
  is debian version stretch
  is a HypriotOS
  has a HYPRIOT_OS= entry
  has a HYPRIOT_OS_VERSION= entry
  has a HYPRIOT_DEVICE= entry
  has a HYPRIOT_IMAGE_VERSION= entry
  is for architecure 'HYPRIOT_OS="HypriotOS/armhf"'
  is for device 'HYPRIOT_DEVICE="Raspberry Pi"'
  uses os-rootfs version 'HYPRIOT_OS_VERSION="v2.0.1"'

Finished in 40.25 seconds (files took 0.72588 seconds to load)
17 examples, 0 failures

Finished: SUCCESS
```
So this is already implemented using Ruby! :-)

It could be used to setup Travis, based on the included [CircleCi configuration](https://github.com/publiclab/image-builder-rpi/blob/master/.circleci/config.yml).

icarito, 2018-07-24:
The current tests are [here](https://github.com/publiclab/image-builder-rpi/tree/master/builder/test).

jywarren, 2018-07-24:
yay!!!!!