publiclab/pi-builder #5: Planning issue!

jywarren, 2018-07-12:
a list of issues, one issue per "script" we hope to have the pi-image generator compile, like:

* [ ] take a picture with `raspistill` once per 2 seconds and store in easy to find folder like `DCIM`
* [ ] take a picture when GPIO button is pressed?

* [ ] and a folder in the repo where we can keep these scripts?

jywarren, 2018-07-12:
More here: https://github.com/publiclab/image-builder-rpi/blob/master/publiclab-virtual-pi/README.md

icarito, 2018-07-23:
I setup an automated build process and was able to preload required debian packages.
Setting up networking as per the templates, I managed to build an [image](https://jenkins.laboratoriopublico.org/job/Raspberry%20Kit%20Image/ws/milestone_wifi_hotspot/), that starts a wifi network as access point.

The usb OTG interface is proving useful for debugging the wireless interface.
I found a good reference for a clean implementation.

I've also begun with the camera integration work.

jywarren, 2018-07-23:
> Setting up networking as per the templates, I managed to build an image

amazing!!!!!!! Very cool. Following with great interest!

jywarren, 2018-10-16:
Oh cool, is the OTG interface in the current master branch build?

jywarren, 2018-10-16:
We could add it to the wifi access point info screen too, as an alternative.