publiclab/pi-builder #7: Build recipe template and recommended workflow

jywarren, 2018-07-30:
Now that we have these building, which is amazing, let's think about how we might make this a quick and standard workflow. I'm imagining different people might want to prepare an `.img` from the following inputs:

* a build script, not necessarily included in this repository?
* a starting image to build on (although for now Raspbian is fine, so let's not think too hard about this yet)

Additionally:

* a wifi hotspot name and pwd, and whether to enable the "captive welcome screen"
* (optional for the narrower case of WiFi hotspot-based applications) - a "captive welcome screen" with
    * link(s) to the installed applications/apps

... and get these outputs:

* an `.img` built from the build script

What would be the ideal workflow from start to finish to specify these things? Maybe what's the current workflow and what would we like to work towards?

Thanks!!!

jywarren, 2018-07-30:
@icarito what do you think? We don't have to do everything at once, but trying to imagine what our ideal goal is...

icarito, 2018-08-02:
Hi @jywarren I've given this a lot of thought and some work!

I can see others would like to build the image with custom settings, and they may not have a Jenkins server or a Docker host at hand.

The best solution I could find is setting up the project to be built with Gitlab-CI.
This way, when there's a PR, the CI pipeline will automatically build an image and **it will be available for download**. Users could work within a PR or fork their own repository (in Gitlab) to build their own images.

With regard to *how* to customize, the recommended way to do it is leveraging `cloudinit` [(documentation)](https://cloudinit.readthedocs.io/en/0.7.9/). This is the way Hypriot project does it and it makes a lot of sense. Basically you have to edit [builder/files/boot/user-data](builder/files/boot/user-data) either at build time or after flashing but *before* first boot. Files can be added to the image at the **builder/files/** path.

jywarren, 2018-08-02:
`cloudinit` sounds great. I think having PRs open as examples of how to create a script would be great. So the PRs would not have to actually be merged to generate images, right?

icarito, 2018-08-02:
On 02/08/18 13:50, Jeffrey Warren wrote:
>
> |cloudinit| sounds great. I think having PRs open as examples of how
> to create a script would be great.
>
+1 That's a straightforward workflow!
>
> So the PRs would not have to actually be merged to generate images, right?
>
Exactly! :-)

jywarren, 2018-08-02:
one other idea I had was to have a subfile required in, if that's possible
-- so that the really long script could include all the more complex stuff,
but a simpler subsection with just a simple "recipe" could be a stand-alone
file. Do you think that would be possible (eventually)?

jywarren, 2018-10-16:
Added some extra docs in the README, but we can link to the blog post when we're ready with that too!

https://github.com/publiclab/image-builder-rpi/blob/master/README.md

commit 1fd71984 (2018-07-27): Setup Captive Portal (fixes #7)