publiclab/pi-builder #19: Make button to open in a browser tab

jywarren, 2018-09-24:
This was great! But it kept opening in the popup, which doesn't let you do that much else (like switching to another window). 

[image]

https://github.com/publiclab/image-builder-rpi/blob/742bcf215220dfe272b165e66c3f2ff9722cefeb/builder/files/var/www/index.html#L17

Can we either add a button that says "open in a browser window" (and how would we do that?) AND/OR have a button that says "OK" and closes the popup (but directs you somehow to the right URL?)

jywarren, 2018-09-24:
This may be OS dependent too -- this says it can be done in iOS with `window.open()`, but not in Android:

https://stackoverflow.com/questions/37777989/open-browser-in-android-captive-portal-after-sign-in

More docs: https://en.wikipedia.org/wiki/Captive_portal

icarito, 2018-09-25:
I've not seen this behaviour, I guess it's specific to Macs! It probably stems from figuring out it's in a captive portal. Firefox also warns that it thinks one needs to "login" to this network but there's no such thing. We could try to have links target a new tab.

jywarren, 2018-09-25:
Ah, ok, i wonder if it's as easy as saying "If this is in a pop-up, you can
close it and open a new browser tab." -- would that work?

On a Mac it kind of works - you can close the window but if you navigate to
any URL it doesn't redirect you to http://pi.local/. But we could say "If
this is in a pop-up, you can close it and navigate to `http://pi.local` in
a new browser tab."

Let's do this as a short-term fix. I'll just initiate this edit now, but
for a longer-term fix, is it possible to redirect ALL domains to
http://pi.local/ while you're connected?

jywarren, 2018-09-25:
Cool, did this: https://github.com/publiclab/image-builder-rpi/commit/35826c359cffe7bb74321e3748f8bf4fb220dccd

jywarren, 2018-09-25:
Ah, additional issue. Android phones (not sure about iPhones) don't have `zeroconf`, so they can't access `.local` URLs. So the "open this in a browser tab" doesn't work for them. 

Two ways forward: 

1. We may have to hide that message on Android (we could detect using user-agent strings maybe)
2. We could have the message display the IP address as a fallback.

if we can redirect all requests that'd be even better, but this might not work on Android which'll have a cell internet connection. Hmm.

icarito, 2018-09-25:
Hi,
A few clarifications:

**"is it possible to redirect ALL domains to http://pi.local/ while you're connected?"**

All domains **are** being redirected... the issue you are seeing with some websites is that in this day and age of SSL everywhere, websites where your browser to default to HTTPS will *not work*.

See the wikipedia article you [linked to](https://en.wikipedia.org/wiki/Captive_portal#Limitations):
> _Similarly, as HTTPS connections cannot be redirected (at least not without triggering security warnings), a browser that only attempts to access secure websites before being authorised by the captive portal will see those attempts fail without explanation (the usual symptom is that the intended website appears to be down or inaccessible)._

**But we could say "If this is in a pop-up, you can close it and navigate to `http://pi.local` in a new browser tab."**

The thing is we don't know under what circumstances we'd be in a popup. It appears operating systems and even some browsers like to do special, undefined things if they think they are in a captive portal.

_Another option_ to avoid the pop-up would be to _not_ redirect every domain or see if there's a standard way to signal clients that the network is not intended as a gateway - **or** if we add the gateway functionality perhaps the pop up goes away.

**Ah, additional issue. Android phones (not sure about iPhones) don't have zeroconf, so they can't access .local URLs.**

Actually, since the Pi controls the DNS in the first place, for devices connected to it, .local resolves to the Pi address. The exact mechanism is that _lightttpd_ responds with a redirect to **pi.local** for every domain but **pi.local** - where it presents the files under `/var/www`.

If you're not being redirected to **pi.local** then it's likely you're only trying with HTTPS (see above).

jywarren, 2018-09-25:
thanks for this amazing clarity!

I was getting a 'not found' for http://pi.local (from my browser, not from
the server) -- but can try again to see exactly what kind of error it was.

I'll try again to confirm the redirect works with http://

Thanks!!!

jywarren, 2018-09-25:
In Android i see:

```
Server not found

Firefox can't find the server at www.pi.local.

    Check the address for typing errors such as ww.example.com instead of www.example.com

If you are unable to load any pages, check your device’s data or Wi-Fi connection. 
```

icarito, 2018-09-25:
In my experience Android tends to try to switch wifi networks when it
logs into a wifi without Internet. Check that your Android device
actually is connected to 00-PiCamera wifi network. I checked this in
Android 5 both with Firefox and Chromium!

jywarren, 2018-09-25:
Aha it was connected but I had to turn off mobile data to get the http://pi.local to work:

jywarren, 2018-09-25:
[image]