publiclab/pi-builder #22: Recipe for timelapse with Raspberry Pi camera

jywarren, 2018-09-24:
There are some great tutorials/code snippets for this:

https://learn.adafruit.com/raspberry-pi-wearable-time-lapse-camera/software

https://projects.raspberrypi.org/en/projects/timelapse-setup

icarito, 2018-10-13:
Here's my take on it [timelapse.sh](https://github.com/publiclab/image-builder-rpi/blob/docs/builder/scripts/timelapse.sh). It is extremely minimalistic.

In this case it's taking advantage of RPi-WebCam to signal taking a picture.
Without RPI-WebCam it could call the `raspistill` program. RPI-WebCam conveniently handles file naming, otherwise we'd have had to handle it ourselves.

jywarren, 2018-10-13:
Ooohhhh exciting!

jywarren, 2018-10-17:
Also great because it might be runnable simultaneously with rpi cam? Which
otherwise would be impossible as only one program can use the cam at a
time? Is this correct?

Thanks!

icarito, 2018-10-24:
Yes, correct, with this approach it can work in tandem with the RPi Webcam software.