publiclab/pi-builder #23: Ability to insert random number or string into config files like WiFi

jywarren, 2018-09-24:
This could be useful in setting up a whole set of different Raspberry Pis to be used in the same place -- so the WiFi networks won't overlap. 

The current default is `00-PiCamera` but we could have this change to `00-PiCamera-XXX` where `XXX` is a random string. 

Maybe this could be a simple shell script to create an environment variable as `$RAND_STRING` or something?

icarito, 2018-10-13:
Here's a recipe for a random 6-char string in bash:

    RAND_STRING=`date | md5sum | cut -c 1-6`

We would need to run sed at build time to change relevant files in `builder/files/etc/`.

jywarren, 2018-10-17:
What if we just add this as default to the whole repo, i.e. merge a PR.
Then any different recipe (represented in a PR) would be able to use this
in their configurations. It could also be an initial step towards resolving
this and we could follow up with a change to other config files.

icarito, 2018-10-20:
Yeah the thing is I'm not sure what to implement because there are several levels to this. 
For one thing there's the hostnames which will interfere if we rely on mdns/zeroconf - so we need unique hostnames but then they'll be unpredictable - then there's the wifi hotspot.

One could think to do this at first boot, *but* it will be nearly impossible to get a random bit on first boot on identical hardware without networking or a RTC (unless I'm wrong on that!).

This requires some research!

jywarren, 2018-10-20:
Hmm, interesting problem! I assume with no microphone or mouse input
there's no ready source of randomness... If we do the wifi ssid could we
poll some networking before we initiate the random number generation?

jywarren, 2019-06-21:
How about we decide to make a script that uses the system clock to add 2 random alphanumerics to the end of the WiFi network name. This way the URL stays the same, but we would get `00-PiCamera-XX` if you press that button. This way we get good pseudorandomness (timestamp the button is pressed) and it's optional on all devices.