publiclab/pi-builder #24: Updtae README.md: Add Issue link

starkblaze01, 2018-10-03:
Hi, this is a first-timers-only issue. This means we've worked to make it more legible to folks who either haven't contributed to our codebase before or even folks who haven't contributed to open source before.

If that's you, we're interested in helping you take the first step and can answer questions and help you out as you do. Note that we're especially interested in contributions from people from groups underrepresented in free and open source software!

We know that the process of creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, consider leaving this one for someone new, and looking through our general help wanted issues. Thanks!

🤔 What you will need to know.
Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

📋 Step by Step
  🙋 Claim this issue: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

  📝 Update the file app/views/dashboard/_sidebar.html.erb in the plots2 repository (press the little pen Icon) and edit the line as shown below.

See this page for some help in taking your first steps!

Below it is showing ( a -) which lines to remove, and ( a +) which lines to add:

-  Check planning issue #5 for other plans.
+ Check planning issue [5](https://github.com/publiclab/image-builder-rpi/issues/5) for other plans.

 💾 Commit your changes

  🔀 Start a Pull Request. There are two ways how you can start a pull request:

If you are familiar with the terminal or would like to learn it, here is a great tutorial on how to send a pull request using the terminal.

You can also edit files directly in your browser and open a pull request from there.

  🏁 Done Ask in comments for a review :)
🤔❓ Questions?
Leave a comment below!

Is someone else already working on this?
We encourage you to link to this issue by mentioning the issue # in your pull request, so we can see if someone's already started on it. If someone seems stuck, offer them some help! Otherwise, take a look at some other issues you can help with. Thanks!

jywarren, 2018-10-03:
Hi, great!!! Thank you!

anabellaspinelli, 2018-10-03:
Hey! I can do this one :)

anabellaspinelli, 2018-10-03:
Alright I just went ahead and submitted the PR https://github.com/publiclab/image-builder-rpi/pull/25 😊 

I hope that's ok. Thank you for the thorough guide ❤️

jywarren, 2018-10-03:
please go ahead! thank you!

jamesrbedwell, 2018-10-18:
Hey, is it possible to do this please?

jywarren, 2018-10-20:
Hi, I believe this was solved in #25 -- but we'd love your help with another issue at https://code.publiclab.org!

closed 2018-10-20 by jywarren