publiclab/pi-builder #38: Camera version issue?

jywarren, 2018-10-23:
Hi, @icarito -- should this system work with either version of the Raspberry Pi camera? I've plugged in a v2 camera and don't seem to be able to see through it. Is it possible it needs a slightly different configuration, or should I keep trying things like tightening cables, trying another SD card, etc?

Thanks!

icarito, 2018-10-24:
Yes! I took special care of this by realizing that RPi Web Camera provides a means to issue commands to the Web Camera software by using a special FIFO file. So although only one process can control the 
Camera at one time, the Web Camera software can be controlled by a secondary command. See  https://github.com/publiclab/pi-builder/pull/29/files#diff-5d9c6e9bb13a65af467247a23079fcc8R3 for the command. Some basic docs are here https://elinux.org/RPi-Cam-Web-Interface#Control_Flow

jywarren, 2018-10-24:
Oh, i'm sorry - i meant, there are 2 editions of the camera board itself, and we've mostly tested on the initial version, but I just had some trouble with the 2nd version of the camera hardware. Are you aware of anything we'd have to change to get the more recent version of the camera working? Or maybe I misunderstand your response? I'm not able to see through it at all with this v2 camera. Thanks!

icarito, 2018-10-24:
Oh sorry, I misread the issue. I'm not aware different instructions for different Pi Camera boards. I'm going to look for this and see if there's any difference in configuration that we'd need to set. I've not seen any notice or warning about different hardware versions before - I'd have assumed they'd be the same.

icarito, 2018-10-24:
You could try with `raspistill` in the command line? (it will only work after killing raspimjpeg process). I believe it **should work** without modifications so if you have doubts do check your cables / hardware.

jywarren, 2018-10-24:
OK, weird -- i swapped for a v1 camera and it worked, then back to a v2 and it worked this time! Very odd... i'd tested 3 times on different SD cards, waiting a while and everything. Will keep testing! Maybe i inserted the ribbon cable wrong?

jywarren, 2018-10-28:
Hm, ok, so i ran it on the commandline and got a no memory error:

```
$ raspistill -t 2000 -o image.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly
```

I also tried increasing the memory by changing `gpu_mem=128` to `gpu_mem=256` in `/boot/config.txt` but that didn't work. i'm going to try a few other values, based on https://raspberrypi.stackexchange.com/questions/75537/unable-to-set-gpu-memory-split-on-raspberry-pi-zero-w#75581

I'm wondering if it's because we're running a number of other things at the same time and we're running short on memory. And if the smaller camera doesn't require as much because it simply needs to store fewer pixels during operations.

jywarren, 2018-10-28:
I'm reaching the conclusion that the 2 camera boards I have may be faulty - i did buy them through a 3rd party seller on Amazon. I'll try ordering 2 new ones.

jywarren, 2018-10-28:
Third camera worked! Whew! I think it may actually have been hardware issues on the others. I will try the 2 new ones to double check, but this is now working!

closed 2018-10-28 by jywarren