publiclab/pi-builder #41: WiFi captive portal cross-platform access issues

jywarren, 2018-10-29:
Different platforms treat the "captive portal" differently:

1. iOS pops up a notification with the welcome page in a webView mini browser, and you can see through the camera, but if you dismiss the window, it disconnects from the network
2. Android works fine with the welcome page in the prompt window (similarly a kind of mini browser instance) but Chrome and Firefox can't access http://pi.local even though it doesn't connect
3. Mac OS pops up a mini browser window with the welcome page, and you can escape out or close the popup and use Chrome or another browser to open http://pi.local manually, but it's not redirected-to

I think as we collect information about these scenarios, we can refine the exact captive portal settings and add other solutions!

jywarren, 2018-11-01:
Is our approach using the 204 http response code as in the Wikipedia article?

jywarren, 2018-11-01:
Figured out that I have to turn off mobile/cellular internet on my Android (Pie) device to be able to load http://pi.local, then it works perfectly!

jywarren, 2018-11-02:
iOS seems to really reject WiFi access points that have no real internet connection: https://serverfault.com/questions/709909/wifi-access-point-without-internet-access-ios-devices-disconnect

These folks have found you can enter a static IP and a few other configs on the iOS device, and get it to work: https://discussions.apple.com/thread/7705927 although that seems like a not very ideal solution...

jywarren, 2018-11-02:
And it says:

> Roland (maker of some very high-end, expensive mixers) gets around this because their wifi network is identified as a "device" rather than an actual network. Connecting to their mixer is only a matter of answering "ok" to the question about connecting to that device, even without an internet, and then I'm good.

jywarren, 2018-11-15:
Kind of cool, here's the message Chrome shows for Amtrak's captive portal:
[image]

minanagehsalalma, 2020-11-03:
@jywarren  hi mate were you able to solve it ?!