publiclab/pi-builder #44: Installing Witty Pi energy manager software

imvectech, 2018-11-01:
Hi all. We've been trying to install Witty pi software and we don't exaclty know how to make a pull request. We are unable to connect to internet using the combined image turning off the ap0 default acces point. 
This script installs the Witty Pi energy manager software (done on a raspbian) plus two tricks we arrived to after fighting a lot with it (one realted to the gpio4 and the other the lack of a scheadule.wpi file) . First time we write a script so maybe is quiet a bit rough ;D (we don't even know if sudo su is necessary) ;)

 echo "Installing Witty pi mini packages"
sudo su
mkdir /home/wittypi
git clone https://github.com/uugear/Witty-Pi-2.git /home/wittypi
chmod +1 /home/wittypi/installWittyPi.sh
./home/wittypi/installWittyPi.sh
wget https://gitlab.com/imvec/anoiacam/blob/8dc8b9c798d2d6c79ec4267b90b8175fd84c7682/schedule.wpi /home/wittypi/wittyPi
rm -rf /home/wittypi/wittyPi/daemon.sh
wget https://gitlab.com/imvec/anoiacam/blob/11d387a98e0283f91e9ae7118a6456a4360caa7c/daemon.sh /home/wittypi/wittyPi

[wittypi.txt](https://github.com/publiclab/pi-builder/files/2539718/wittypi.txt)

jywarren, 2018-11-01:
Super! So you can click the ✏️ next to the top right on this file, and begin adding your content -- probably near the bottom:

https://github.com/publiclab/pi-builder/blob/master/builder/chroot-script.sh

We can debug as it builds and make necessary adjustments gradually! Awesome! 🎈 📷 ⚡️

jywarren, 2018-11-01:
Excited so I started us off here: https://github.com/publiclab/pi-builder/pull/45

jywarren, 2018-11-01:
...it's building!!!!

imvectech, 2018-11-01:
O_o great!!

jywarren, 2018-11-01:
I'm not sure, i think i did that here? https://github.com/publiclab/pi-builder/pull/45/files

jywarren, 2018-11-01:
and if you want you can open your own PR with this so you can keep modifying it. I usually do the modifications in the online editor, actually!

imvectech, 2018-11-01:
Yeah there it is! So in a while a new image would be ready?¿

imvectech, 2018-11-01:
"pipeline failed with gitlab" I did something wrong ;D It says I need "push" acces to edit the file.

jywarren, 2018-11-01:
Yes - there are instructions in the PR to fetch it. We want to add a bot
that pastes in the final link when ready, but haven't figured out how yet.
:-)))

imvectech, 2018-11-01:
Just created my first pull request at https://github.com/publiclab/pi-builder/pull/46 ;D