publiclab/pi-builder #48: Auto-rebase?

jywarren, 2018-11-05:
This app seems to be able to auto-rebase anything labeled `autorebase`, and i'm going to install it and try it on my own PR!

https://github.com/tibdex/autorebase

jywarren, 2018-11-05:
Ok, after accidentally auto-merging a PR i realized this bot will actually do a merge if it's allowed to. So I added branch protection and required a review. I think it's possible we could remove the review requirement and it still wouldn't have permissions to merge to master (i don't think it ever should). 

But in #50 i don't see it auto rebasing...?

jywarren, 2018-11-08:
I'm worried this is stopping builds, so i'm testing turning it off: https://github.com/apps/autorebase

@icarito in case this helps!

icarito, 2018-11-09:
Thanks! I'm afraid I've tried it all and for some reason gitlab is not triggering status updates on github's PRs like it was doing before. It may be because of the project rename, but even re-creating a fresh repo either with the auto-importer or manually didn't succeed:

    https://gitlab.com/publiclab/pi-builder-builder/-/jobs
    https://gitlab.com/publiclab/pi-builder-ci/-/jobs

And the original:

    https://gitlab.com/publiclab/pi-builder/-/jobs

At this point I'm thinking of seeking gitlab support.

jywarren, 2018-11-09:
oh, that sounds frustrating. I think asking GitLab for support sounds like
a good idea!

jywarren, 2018-11-21:
Not sure if we got the build triggering completed, but as we make changes to the base repo, the idea of auto-rebase sounds more and more important. Thinking ahead to this a bit more!