publiclab/pi-builder #63: Testing on different Raspberry Pi models

jywarren, 2019-03-22:
I just tested this out on a Raspberry Pi 3 Model B with a camera, and it worked great, no changes at all to the SD card.

It's already been extensively tested on the Pi Zero W.

Anyone try it on other boards? Know what it took to get it working on others?

@xose maybe?

jywarren, 2019-03-22:
cc @icarito - was pleased about the Pi 3B!

jywarren, 2019-03-27:
Also for boards with no wifi, #28

jtravs, 2021-01-25:
Hi I'm trying to test this on a RPi 3 A+. I can connect to the wifi network, I can also log in via ssh, grab an image with raspistill and download with with scp, so both teh wifi and camera work. But the web interface doesn't seem to work, I just get this

[image]

I get the same output if  I try either http://pi.local/cam or http://pi.local

Am I doing something wrong?

jtravs, 2021-01-25:
OK, my mistake, it works fine if I use the artifact from https://github.com/publiclab/pi-builder/pull/54
I have to admit, it has been pretty hard working out what is the best/latest image to use. But it is excellent now it is working! Thank you!